psas/liquid-engine-test-stand #2: Identify test stand communication interface

matthewhauer, 2016-06-24:
A point that tumbled out of email'ing Nate:

How are the sensors and valve actuators connected to the outside world/interface?  How are the data signals and command signals bouncing from software to hardware and back?

This ticket shall be accepted when the interface has been identified adequately to make software microcontroller requirement decisions from.

7deeptide, 2016-06-27:
AFAIK, this will be achieved by a combination of COTS microcontrollers (e.g. beaglebone or somesuch) and our own open-source marionette DAQ board [http://marionette-daq.github.io/](url). Andrew and/or Gavin owns this issue.

matthewhauer, 2016-07-06:
https://marionette-daq.github.io/ ... the non-https page 404's

Can we get Andrew and/or Gavin tagged on this issue?

matthewhauer, 2016-07-06:
Note to Matt: ask Andrew and/or Gavin about using the LTC3 as a LFETS controller during testing.

This will additionally answer the question of interfacing with the hardware, because the LTC3 will use psas_packet to exchange data from the flight* hardware to the user interfaces.

LateralGs, 2016-07-06:
Have we identified what the control loops in the system are and if they are safety critical or real time in nature?

matthewhauer, 2016-07-06:
@LateralGs 

>Have we identified what the control loops in the system are and if they are safety critical or real time in nature?

I bounced this off Erin last night, who said "Yes and yes".  I opened #6 since PSAS/LFETS/LTC probably need a longer answer, and we should have the answer to that question somewhere publicly visible.

matthewhauer, 2016-07-06:
I talked about this issue with Andrew last night.

The instrumentation on the test stand will be plugged into a Marionette DAQ board.  See https://marionette-daq.github.io/ for more information.

The DAQ board will either be plugged into the LTC via USB, or into a testing laptop via USB.  "A UNIX based system" [Andrew], the details are flexible.  If plugged into a laptop directly, the interface shall look for the two serial ports that Marionette exposes, pull data from one, and push commands to the other.  If plugged into LTC, the LTC is expected to broadcast telemetry data, that a software dashboard can pick up.  Or other telemetry monitors.  This telemetry data is expected to be some form of psas_packet.

In return, commands are expected to use a psas_packet command type.

Both packet types are currently unspecified.  The LFETS telemetry is likely to be a new packet type, since the existing descriptions do not specify pressure information.  The command packet type may be a new type, or may be a modification to an existing command format.  Either way, this is a topic to bring up with @jameysharp !

So.  Hardware -Cu- Marionette -USB- {?LTC -wifi-} Dashboard from layout.  The test stand shall communicate either with the Marionette DAQ directly, or with psas_packet telemetry messages.

The contents of both the telemetry and the command messages are yet unspecified.  (which means MOAR TIKKETZ)

matthewhauer, 2016-07-06:
Ticket accepted with https://github.com/psas/liquid-engine-test-stand/issues/2#issuecomment-230835335

Microcontroller requirements: Marionette DAQ, with python library to access exposed 'serial port' connections to the board.
Additionally, the data from Marionette shall be packaged up in a psas_packet telemetry message.

Followup tickets: #9 and #10

closed 2016-07-06 by matthewhauer