psas/liquid-engine-test-stand #8: Encourage small frequent commits

matthewhauer, 2016-07-06:
Identify ways to encourage LFETS members to commit and push changes to the repository more often than at the end of the term.

"Small, frequent, well-described commits are MUCH easier to perform than monolithic commits.  They also have the added benefit of sharing your changes with the team, with having an off-site backup, and having change tracking."  -Matt

This ticket shall be accepted when an effective method of encouraging git commits/pushes is identified.

matthewhauer, 2016-07-06:
I'm not sure if it'd encourage commits, but declaring that project files should be exchanged by pushing to and pulling from GitHub, and discouraging emailing project files between members, may nudge everyone in 'the right direction'.

Or students will be students and do what's easy, which would be to continue to email files.  TBD.

IanOsgood, 2016-07-06:
> declaring that project files should be exchanged by pushing to and pulling from GitHub, and discouraging emailing project files between members

That is absolutely the right thing to do. Teach these students good habits now please; their future employers and future generations of PSAS students will thank you for it!

Also, if they are worried about committing things before they have been tested, they should be taught how to use branches in git for experimental work.  Or even how to make personal forks of the PSAS repositories from which pull requests can be made.  Perhaps you should have a git seminar once a quarter to educate new students about these kind of development techniques.

matthewhauer, 2016-07-06:
> That is absolutely the right thing to do. Teach these students good habits now please; their future employers and future generations of PSAS students will thank you for it!

> Also, if they are worried about committing things before they have been tested, they should be taught how to use branches in git for experimental work. Or even how to make personal forks of the PSAS [...]

There is a LOT of non-software material that gets posted to the PSAS github repos -- Jupityr (sp) notebooks, some with code and others without, technical schematics, SolidWorks files, launch data and analysis, etc. etc.  Git/VCS is a rough fit with hardware, but still works well enough.  The challenge is in getting the students just to use the system first, rather than all the other informal solutions.

The "bazillions of projects" approach that PSAS employs also presents some organizational challenges, at least when it comes to figuring out what's going on, what's "live" vs stale, and what needs attention.

I'd be game for drafting a "git and tickets" presentation for PSAS if it was useful, though I can't say that I've learned the best habits myself.

closed 2020-09-16 by mwasserm