psas/liquid-engine-test-stand #9: Specify LFETS/LTC command packets

matthewhauer, 2016-07-06:
The LFETS will have both a 'direct' (plug in to Marionette DAQ via USB) and 'telemetered' configuration.  In the latter, a psas_packet type shall be used to send command message to the LFETS.

The contents of these command messages needs to be specified.  In doing so, the available actions on the test stand shall specify the options, and be documented.

This ticket shall be accepted when a psas_packet 'LFETS command' type has been written up.  A follow-up ticket shall be to implement said type.

matthewhauer, 2016-07-06:
Tagging Jamey as psas_packet SME.

matthewhauer, 2016-07-13:
Lean on @KyleMeeuwsen about the list of test stand commands required for the command packets.  IE, the individual commands to define in the packet spec.

jameysharp, 2017-07-07:
Sorry I never got to this, but I'm not sure what I was expected to do. If you're still interested, can we talk about it at a regular Tuesday evening meeting?

closed 2020-09-16 by mwasserm