psas/liquid-engine-test-stand #12: CAD Layout

matthewhauer, 2016-07-13:
Tyler, Kyle.

Construct a CAD drawing with simplified models of the parts to lay out the structure of the test stand and how they fit together.  Make sure the human-accessible parts are all together.

Simple primitive solids are suitable for this task at this point.

This ticket shall be accepted when a  3D diagram has been posted to the GitHub account.

KyleMeeuwsen, 2016-07-27:
Asa, I have been unsuccessful at moving the adobe file to github. For some reason Github will not take the adobe file. I have not tried the solidworks file and will attempt this either tonight or possibly tomorrow.

asa-grant, 2016-07-27:
Thanks man. Thanks for doing this for me, I need to sit down with github for an hour or so and just familiarize myself with it that way I can do it on my own.

-Asa

7deeptide, 2016-07-27:
Guys, please let me know if you'd ever like a sit-down git tutorial from me... I'd be happy to do it at the next propulsion meeting if you like (you'll just need to bring your laptops).

asa-grant, 2016-07-27:
That would awesome!

Thanks!

-Asa

closed 2017-02-02 by 7deeptide