psas/liquid-engine-test-stand #16: Need a safety manual for the transportation, handling, and installation of all materials and fluids.

KyleMeeuwsen, 2016-07-13:
Examples: Do not place N2 tanks in the back seat of a car with closed windows. If N2 leaks, individuals in the car my asphyxiate.

KyleMeeuwsen, 2016-08-03:
See manual (template) for launch and apply it to test stand. Ensure compliance with ASME, Cryo handling, Boiler Code, etc. Erin to comment more on this.

MechBertrand, 2017-07-03:
Marked similar topics as duplicate. Renames this issue as SOP:... from "Need a safety manual for the transportation, handling, and installation of all materials and fluids."

Link to standardized procedure book: https://github.com/psas/procedure-book

Will soon be creating an LFETS SOP and Operators' Hanbook starting with a copy of this link.

MechBertrand, 2017-07-03:
Closed Duplicates:
https://github.com/psas/liquid-engine-test-stand/issues/13
https://github.com/psas/liquid-engine-test-stand/issues/15

closed 2020-09-16 by mwasserm