psas/liquid-engine-test-stand #17: Determine appropriate valve actuation times

matthewhauer, 2016-07-13:
Ask Bob and ARocket about reasonable times for ball valve actuator for the ethanol and LOX main valves.  We don't want to screw up the combustion chamber with questionably timed LOX and ethanol flow.

This ticket shall be accepted when ...

KyleMeeuwsen, 2016-08-24:
The two second open time on the LOX actuator is acceptable since based on Gregory and Robert's feedback as long as the valve is a ball valve. The ball valve allows us to crack the valve open, allowing a small amount of LOX to flow through the system, eliminating the two phase flow.

KyleMeeuwsen, 2016-08-31:
Selected 3/8" brass ball valve with 6.4 Cv. Asa needs to send the part number and MFG to Kyle.

closed 2016-12-23 by 7deeptide