psas/liquid-engine-test-stand #18: Need burst discs for LOX and Ethanol manifold

KyleMeeuwsen, 2016-07-13:
**Acceptance Criteria:**
4 burst discs from Oseco that are rated to release at 1000 PSI. 
Detailed information about housing and connection fittings.
Releasing to atmosphere --> Must have a protective casing around disc to ensure there are no projectiles.

KyleMeeuwsen, 2016-07-27:
Hello Trevor,

Have you been able to make any progress on this since we last spoke. Were you waiting on some input from me?

KyleMeeuwsen, 2016-08-03:
I checked with Oseco and the quote is substantially larger than the $25 we'd heard prior. For a set of 4 good discs, with scoring to prevent rupture fragmentation, we'd be looking at $227 each ($908 for four with holders). I've put in for quotes with five other companies. Two came back in the same range of prices, a little higher. The other three claim to have quotes Wednesday (Aug 3) for me. 
It's looking like the cost of discs and holders will be significant.
I'll keep looking to see if anything could be reasonably below that cost, otherwise that's the only option.

Best,
Trevor

KyleMeeuwsen, 2016-08-03:
Trevor, 

Erin and I spoke and we were wondering if you wanted to get in touch with Robert Watzlavick. We are wondering if Robert specified a higher burst pressure (allowing for a thicker burst wall, requiring less machining accuracy) or if maybe Oseco is specing this for cryogenic application. I think that is great that you are looking into other companies. You are making great progress. Perhaps Erin Schmidt can provide you with Robert's email. 

Best,
Kyle

TrevorLeake, 2016-08-03:
I didn't speak with Robert Watzlavick, but I will get his email from Erin. Oseco was spec'ing for 0F, though you are right that cost goes up a ton with cryogenic applications. I'll keep talking with other companies to look for a lower cost, and see if Robert can shed any light on the price difference. 

I think it must be precision machining that drives the cost here, cause the material on most discs, and the ones quoted, are 316SS, and this is the bare bones simple version of a rupture disc. 

All the best,
Trevor

KyleMeeuwsen, 2016-08-22:
Hello Trevor,

Any new information on this?

Best,
Kyle

TrevorLeake, 2016-08-22:
Sorry, I've been trouble making any useful findings. Basically Erin's guess was correct; a lot of manufacturers have gear which makes precision machining at the level we need more costly. I've been inquiring with a few of the companies' reps about a cheaper geometry. I am hoping that if I can't find a cheap place for purchase, we can just use a line adapter to a different (I think smaller) diameter... whichever provides easier machining. 

Having the reps' available I figured I could do without bothering Robert, but I will now bother Robert.

Again, sorry for the kindof ridiculous turn-around on what appeared an simple buy. 

Best wishes,
Trevor

KyleMeeuwsen, 2016-08-22:
No worries, thanks for continuing to look into this. This level of difficulty for finding a part at an appropriate cost is pretty standard it seems. Thanks for continuing the search.

Kyle

TrevorLeake, 2016-08-23:
Alright, this can be done cheap. I'll post full details in a couple days when it's all ensured.

KyleMeeuwsen, 2016-08-24:
Awesome, that's really exciting! Thank you for diving in to this.

Best,
Kyle

KyleMeeuwsen, 2016-09-21:
We have replaced the burst discs with a cryo rated pressure relief valve on the LOX side and a normal pressure relief valve on the Ethanol side.

closed 2016-09-21 by KyleMeeuwsen