psas/liquid-engine-test-stand #23: (1) Acquisition Phase: Outsource hydro testing on LOx and Aluminum.

KyleMeeuwsen, 2016-08-03:
Ensure that company understands that we have modified the aluminum tank (drilling through bottom).

taruhhdactyl, 2017-06-15:
The tanks will need to run at about 700 psi, but we need a safety factor. We can probably just have them tested against the DOT standard printed on the tanks.

Below is a list of companies that will do hydro testing for $20 (so we hear) and are aware of the special procedures for galvanized tanks as well as checking the REE # to give a tank a + rating on the hydro.

US Cylinder in Vancouver
Sanderson Fire and Safety downtown Portland - **503-889-3110**
United Fire and Safety on MLK - **503.249.0771** - Tara emailed 6/15

MechBertrand, 2017-06-15:
From [Hydrostatic Testing FAQ](http://www.tymsinc.com/hydrostatic-faq.html)

> All cylinders have ratings that start with “DOT” or “ICC,” followed by the service pressure of the container.  For example “DOT 3AA 1800” means a DOT approved cylinder made to the “3AA” specification rated for 1800 psi.

7deeptide, 2017-06-15:
We have the equipment to do hydro testing ourselves. At this point no need
to outsource it...

closed 2020-09-16 by mwasserm