psas/liquid-engine-test-stand #25: (1) Acquisition Phase: Finish P&ID

KyleMeeuwsen, 2016-08-24:
Incorporate the ignitor system into the P&ID. (Kyle B) will be providing a schematic of the ignitor system and I will add it to the P&ID. 

We will not be adding a wiring diagram since this will be developed by another group.

7deeptide, 2017-04-01:
@JacobTiller @asa-grant  
Issues with the system level PID:
1. Mix of ISO and ASME symbols (MFV & MOV use ASME for instance)
2.Purge system check valves are backwards
3. There will be either an orifice or flow meter on the fuel side just prior to the engine fuel manifold inlet
4. No fuel line to engine

Bigger issue with the system itself: We can not stage the engine startup with this layout. We have to open the main valves before we can start the igniter. Obviously this is problematic if we want to verify good ignition before opening the main valves. Maybe we should move the igniter lines behind the main valves and give them their own purge lines?

JacobTiller, 2017-04-04:
@7deeptide @asa-grant 

1. I believe I have the MFV and MOV valve symbols correct (based on what I know about them at least)
2. Fixed purge valve orientation
3. added a place holder for an orifice or flow meter on the fuel side
4. added the missing fuel line.

As for the bigger issue,  I took a stab at changing the igniter lines and adding a second purge line to the system.   I created an "Alt A" system PID and pushed it.

7deeptide, 2017-04-04:
@JacobTiller @asa-grant The Alt-A schematic is a good start. We don't need to separately purge the igniter and main tubes, they can all be controlled by a single valve (e.g. valve 9).@

Also needing to be added:
1. Heat exchanger coil on the LOX igniter tube.
2. Pressure transducer and K-thermocouple on the pre-injection fuel manifold.

JacobTiller, 2017-06-15:
Closing issue as all of the items on it were completed.  PID has a number of other items still needed so I'm opening a new issue for the new updates needed.

closed 2017-06-15 by JacobTiller