psas/liquid-engine-test-stand #28: (1) Acquisition Phase: Spec and select a Lox Fill Vent Valve

KyleMeeuwsen, 2016-08-31:
From Robert's Page:
Here is the current plumbing layout. The pressurization lines will use 1/4 inch and the liquid lines will use 3/8 inch 6061-T6 aluminum tubing. Igniter lines will be 1/8 inch 3003-O tubing. The LOX vent will be located between the fuel and LOX tanks to keep the cold liquid away from the regulator, pressure transducer, and burst disc.

KyleMeeuwsen, 2016-08-31:
ARocket - Robert and Others:

It seems sticking LOX valves are a recurring problem - CS had issues recently and I've experienced the problem several times myself.  On my recent set of cold tests, I noticed that the first run of the day with LN2 was OK but if I didn't let the exterior of the valve dry out sufficiently between runs, moisture condensed from the atmosphere would find its way into the valve bearing and cause it to freeze up.  It would be a major redesign to add another seal to keep moisture out of the stem/bearing section of the valve body (there's not much room).  I was thinking an inexpensive modification would be to enclose the LOX vent, fill, and main valves in plastic wrap of some sort with small plastic tubes coming out of them.  Then, I could squirt some N2 into them from a central point prior to the test to displace the moisture-laden air.  Since it wouldn't be a completely sealed system, some air would leak back in but it seems like it would be an improvement over nothing.  Clear PTFE bags are available but I'd need some way to seal the openings.  Wide (1 inch or greater) PTFE tape might work also, just wrapping it around a bunch of times.  Kitchen wrap would also work but I'm not sure about whether it would crack when cold or whether it is oxygen compatible.

over the seams? Or stainless foil, like the sort of stuff used for
keeping out oxygen when heat treating small parts? Or you could do the
whole covering out of foil; I'm not sure which option would be easier
in practice.

Aluminum foil seems like a better idea - thanks.  It doesn't need to be completely airtight - in fact, it probably needs a little drain hole at the bottom in case there is LOX leak so it doesn't fill up.  And foil vs. PTFE wrap would allow the servo to survive slightly longer (although not much) in case of a fire.

Also be aware of "cryopumping"--cold surfaces cool the air around them, causing a drop in pressure and sucking in more air. I'm not sure if you were planning on just a "static" volume of GN2 around your valve (not an active purge). If so, it's possible moist air could still be sucked into the enclosure. How much and whether it would be a problem, I don't know.

KyleMeeuwsen, 2016-08-31:
3/8" brass piloted solenoid valve with *all* PTFE seals and correct-ish pressure rating will likely work, if you can survive the weight. Needs to be cleaned for oxygen service. You will need to buy or make an appropriate "spanner wrench" to pull the pilot stem assembly. 

For a 4 gallon tank a 1/4" NPT valve with 3/8 tube adapters would likely even be sufficient. Piloted valves have higher pressure ratings for the same amount of Cv compared to direct acting, and you don't need the super low latency of a direct acting valve. A piloted solenoid valve is still faster than a pneumatic ball valve or electric plug valve. 

Make sure it stays dry inside. Don't insulate the valve, you don't want the coil to get too cold because the casing is typically plastic.

KyleMeeuwsen, 2016-08-31:
# Requirements:
Not sure if we need a high Cv since this will be releasing evaporated Oxygen at a low pressure differential during LOx filling. 
Need a valve rated for ~1000 PSI for operating pressure. 
Required - All PTFE seals. 
3/8" NPT or Ferrule fittings on both ends
Piloted Solenoid valve

KyleMeeuwsen, 2016-09-06:
For cryogenic rated valves at 1/2", estimated cost is around $800, and max differential pressure typically maximum at ambient temp and around 100 PSIG...

asa-grant, 2016-09-06:
Agreed, with the exception of valves with prohibitively small orifice sizes, I cannot find a normally open cryo-rated valve that can handle pressures anywhere near our working pressure

-Asa

KyleMeeuwsen, 2016-09-07:
Swagelok does not offer straight thread and NPT is not acceptable since NPT will require PTFE which cannot be used for LOX and makes disassembly difficult. Thus we are selecting SS-45S8 with a 12Cv and Ferrule fittings.

closed 2016-09-07 by KyleMeeuwsen