psas/liquid-engine-test-stand #37: Prop-tank in-line pressurant diffusers

7deeptide, 2017-02-02:
Pressurant diffusers are needed on both the LOX and IPA tanks to minimize the amount N2 mixing and diffusion that occurs in the tank (this especially important with LOX).

Diffusers need to be designed and fabricated.

For reference/inspiration:
[](https://www.nasa.gov/sls/sls-diffuser.html)
[](https://ntrs.nasa.gov/archive/nasa/casi.ntrs.nasa.gov/19660000341.pdf)

IanZabel, 2017-05-05:
I'm ready to get going on it, but I want to ensure I have the right dimensions/parameters. From the parameters in [the math section](https://github.com/psas/liquid-engine-test-stand/blob/master/analysis/feed_system_calculations.ipynb), you've got the mass flow rates for both the LOX and IPA tanks as well as tube diameters [(where gas flow is labeled?)](https://ntrs.nasa.gov/archive/nasa/casi.ntrs.nasa.gov/19660000341.pdf).

- What were the threads required, using the NASA example linked as a reference (just past the flange)?
- Am I referencing the correct mass flow rates and pressures?
- Any specific requirements for the flange/plug?

With that I can start on the CAD and whip something up.

CStarker, 2020-03-29:
Is this still necessary or should this be marked as complete?

closed 2020-09-16 by mwasserm